morrownr/USB-WiFi #621: Netgear AX8000 Working Fedora linux

Juju8901, 2025-06-05:
Tried quite a few including a few on the plug and play but for some reason they switched over to realtek but kept the brand and model as you said would happen. Picked up a Netgear and everything was automatically plug and play! had a great time.

Running Fedora 42 Workstation on a Framework 13

Linux oxtail 6.14.9-300.fc42.x86_64 #1
Thu May 29 14:27:53 UTC 2025 x86_64 GNU/Linux

morrownr, 2025-06-05:
Hi @Juju8901 

Glad you found an adapter that you like.

> Tried quite a few including a few on the plug and play but for some reason they switched over to realtek but kept the brand and model as you said would happen.

I try to avoid those situations in The Plug and Play List. If you can show me the links that are going to a product that has been changed out, please let me know so I can make changes.

Juju8901, 2025-06-05:
@morrownr 

These did not work because of realtek
https://www.bestbuy.com/site/tp-link-archer-tbe6500u-tri-band-be6500-6-5-gbps-wi-fi-7-wireless-ultra-fast-usb-3-0-adapter-with-advanced-wpa3-security-gray/6623611.p?skuId=6623611

https://www.bestbuy.com/site/tp-link-axe5400-wi-fi-6e-high-gain-wireless-usb-adapter-black/6578372.p?skuId=6578372

https://www.bestbuy.com/site/asus-usb-be92-nano-be6500-tri-band-wi-fi-7-usb-network-adapter-black/6594275.p?skuId=6594275

this one did work:

https://www.bestbuy.com/site/netgear-nighthawk-axe3000-tri-band-wi-fi-6e-usb-3-0-adapter-black/6520471.p?skuId=6520471