morrownr/USB-WiFi #627: [Help]: List of chipsets with mesh point & 6 GHz band capability

pwnept, 2025-06-25:
### Checklist

- [x] I acknowledge that support is provided on a best-effort basis.
- [x] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

Linux OpenWrt 6.1.57 #0 SMP Wed Oct 16 11:35:31 CST 2024 aarch64 GNU/Linux

### lsusb

Bus 002 Device 001: ID 1d6b:0003 Linux 6.1.57 xhci-hcd xHCI Host Controller Bus 004 Device 001: ID 1d6b:0001 Linux 6.1.57 ohci_hcd Generic Platform OHCI controller Bus 006 Device 001: ID 1d6b:0001 Linux 6.1.57 ohci_hcd Generic Platform OHCI controller Bus 005 Device 002: ID 04ca:3802 MediaTek Inc. Wireless_Device Bus 001 Device 001: ID 1d6b:0002 Linux 6.1.57 xhci-hcd xHCI Host Controller Bus 003 Device 001: ID 1d6b:0002 Linux 6.1.57 ehci_hcd EHCI Host Controller Bus 005 Device 001: ID 1d6b:0002 Linux 6.1.57 ehci_hcd EHCI Host Controller

### rfkill

(no rfkill)

### dkms

```text
(no dkms)
```

### iw

```text
phy#0
        Interface wlan0
                ifindex 15
                wdev 0x2
                addr 74:4c:a1:8c:bb:6f
                ssid aisfibre_5G
                type managed
                channel 100 (5500 MHz), width: 80 MHz, center1: 5530 MHz
                txpower 3.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       0       0       0       0       0       0               0
```

### What happened?

Hi, thanks for the helpful documentation in this repo.

I'm trying to find USB WiFi adapters that support both:
1. Mesh point capability (i.e., Supported interface modes: mesh point from iw list)
1. 6 GHz band (e.g., for WiFi 6E or 7)

I’ve looked through the [USB_WiFi_Chipsets.md](https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Chipsets.md) page which lists some adapters with WiFi 6E support, but if I understand correctly:
- Some WiFi 7 adapters do not necessarily support 6 GHz.
- Even if a chipset supports 6 GHz, the adapter using it might not (e.g., COMFAST CF-952AX / CF-953AX with MT7921AU reportedly don’t support 6 GHz).
- Similarly, support for mesh point mode seems to vary and is not always documented clearly.

Is there:
- A known list of chipsets or USB adapters that support mesh point mode?
- A more detailed compilation of adapters’ iw list output (especially Supported interface modes: and supported bands)?

Any help would be much appreciated!

pwnept, 2025-07-09:
Would roaming be more applicable to Wifi 6/6E/7 capable USB-WiFi cards as opposed to mesh? I'd have to put in a bit more work configuring the VMs' networks but then maybe hardware choices would be easier to find?

morrownr, 2025-07-09:
I cannot answer your question because I do not know what problem you are trying to solve.

nerdalertdk, 2025-10-15:
> Hi [@pwnept](https://github.com/pwnept)
> 
> ```
> I'm trying to find USB WiFi adapters that support both:
> 
> 1. Mesh point capability (i.e., Supported interface modes: mesh point from iw list)
> 2. 6 GHz band (e.g., for WiFi 6E or 7)
> ```
> 
> I have USB WiFi adapters with nearly all modern (WiFi 5 and later) usb wifi chips. I say with a great deal of confidence that what you are asking for does not exist. I am not aware of any Realtek chips/drivers that have ever supported Mesh Point. There are some Mediatek chips/drivers that support Mesh Point but I think they are WiFi 5 and earlier. I would have to investigate to be sure and be specific.
> 
> This is not a common request. Maybe if you explain what you are trying to accomplish, I can offer some ideas.

Hi

I'm also looking in to doing mesh point with Realtek chips, RTL8192
and so far I've come to the same conclusion as you that mesh and rtl driver don't work.

But when looking at my card I do see it has support for mesh point frame types, so is the lacking of mesh support just a driver issue or hardware limit ?

Ps. only need a mesh card with wifi5 support :)

Cheers Michael