morrownr/USB-WiFi #631: [Help]: Increase txpower to 30. Stuck at 20 - PAU0D / mt76

trendxme, 2025-07-08:
### Checklist

- [x] I acknowledge that support is provided on a best-effort basis.
- [x] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

6.11.0-29-generic #29-Ubuntu SMP PREEMPT_DYNAMIC

### lsusb

Bus 002 Device 015: ID 0e8d:7612 MediaTek Inc. MT7612U 802.11a/b/g/n/ac Wireless Adapter

### rfkill

n/a

### dkms

```text
using mainline driver (if there is a custom one that's better, lmk)
```

### iw

```text
phy#16
	Interface wlx9cefd5f809e0
		ifindex 20
		wdev 0x1000000001
		addr .
		ssid .
		type managed
		channel 100 (5500 MHz), width: 80 MHz, center1: 5530 MHz
		txpower 20.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
```

### What happened?

Hi, I was wondering if anyone had any ideas on how to increase the txpower as it's stuck at 20. My AP is a bit far away, but my cell phone has no issues getting high speeds. It seems there are a lot of retries during TX with this adapter.

`sudo iw dev wlx9cefd5f809e0 set txpower fixed 3000` does nothing...

Using debugfs, we can see the kernel driver should support up to 30...
```
a@a:~$ sudo cat /sys/kernel/debug/ieee80211/phy16/mt76/txpower
Target power: 30
     Delta: -2 -2
```

I tried following the kernel driver but don't really understand all the values. Is there an easy way to set it to 30? Worst case I can patch the kernel driver with a new debugfs entry to set the value but I'm afraid the firmware might limit it.  I contacted the support for this adapter but they couldn't help. Might need to return it.

morrownr, 2025-07-08:
Hi @trendxme 

> Is there an easy way to set it to 30?

No.

I need to know what country you are located in. The wireless regdb will have the biggest influence on the questions you are asking.

morrownr, 2025-07-08:
Okay, regdb shows the following for CA:

```
# Sources:
# https://www.ic.gc.ca/eic/site/smt-gst.nsf/vwapj/rss-247-i2-e.pdf/$file/rss-247-i2-e.pdf
# https://www.ic.gc.ca/eic/site/smt-gst.nsf/eng/sf11750.html (6 GHz operation)
country CA: DFS-FCC
	(2402 - 2472 @ 40), (30)
	(5150 - 5250 @ 80), (23), NO-OUTDOOR, AUTO-BW
	(5250 - 5350 @ 80), (24), DFS, AUTO-BW
	(5470 - 5600 @ 80), (24), DFS
	(5650 - 5730 @ 80), (24), DFS
	(5735 - 5835 @ 80), (30)
	(5925 - 7125 @ 320), (12), NO-OUTDOOR
``` 

> (2402 - 2472 @ 40), (30)   and   (5735 - 5835 @ 80), (30)

That means you can go up to 30 on band 1 (2,4 GHz) and channel 149 and above on band 2 (5 GHz)

> (5470 - 5600 @ 80), (24), DFS

For channel 100, the maximum is 24 but that can be reduced by the adapter. Many adapters won't accept tx changes.

I do not have experience with your specific adapter. What you could do is install `wavemon` and tell me what the signal readings are:

For Debian based systems:

Ctrl + Alt + t

$ sudo apt install wavemon

$ wavemon