morrownr/USB-WiFi #641: ALFA AWUS036ACM vs. EDUP EP-AX1672 for WiFi 5 clients

bdantas, 2025-08-15:
Hello again, Nick. Hope all is well. Your page _The_Short_List.md_ has been a golden reference for me over the years. I can't thank you enough for it.

I built my own GNU/Linux router and currently use the ALFA AWUS036ACM for 5 GHz AP. All clients in my household support 802.11ac (WiFi 5). None currently support 802.11ax (WiFi 6). I live in a large home with three floors. AP lives in the most central room on second floor.

I am considering upgrading the router's adapter to EDUP EP-AX1672. But before I go for it I wanted you opinion on whether it is even theoretically possible to achieve a benefit in range, client download speed, and/or client upload speed given that the clients only support 802.11ac. What are your thoughts on this?

morrownr, 2025-08-15:
Hi @bdantas 

Long time, no see.

> Can you think of any theoretical benefits that an EDUP EP-AX1672 AP (with 802.11ax enabled) would provide to 802.11ac clients--compared to ALFA AWUS036ACM?

I do not see any benefits in this situation. Once you have AX capable clients, then it would make a difference. 

Something to be aware for the future is that 6 GHz AP mode is a challenge with adapters and cards in some countries. The US is a good example. The wireless regulations in the US make it where we cannot currently make a 6 GHz AP with an adapter or card.

Stop by anytime.

bdantas, 2025-08-15:
I'll start a new issue with my question about hostapd settings for optimal authentication, since it's a totally separate topic. Sorry for spamming my own thread :)

closed 2025-08-15 by bdantas