morrownr/USB-WiFi #642: hostapd setting recommendations for reliable client authentication

bdantas, 2025-08-15:
Hi, Nick. I have a homebrewed wireless router that uses a usb adapter for creating the AP. I currently use the ALFA AWUS036ACM for 5 GHz 802.11ac AP. Every now and then I'm in a tinkering mood and will try a different adapter just to compare performance (range and client download/upload speeds).

I recently read about the AWUS036ACH adapter here on your github page and was happy to hear that it now has an in-kernel driver. Since I'm still on linux 6.12.11, I could not benefit from the in-kernel driver yet and had to compile the rtw88 driver from its github page. 

After compiling the driver and grabbing the firmware from linux kernel's git, I gave this adapter a spin in AP mode using your hostapd.conf configuration tips (from https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/hostapd-WiFi5.conf). First I tried the foo_capab settings specifically for the rtl8812au chipset with rtw88 driver. Then I just tried the generic settings. Both times, my Android phone could not always authenticate with the AP. I have since returned the adapter for a refund but have been mulling over the problem ever since.

Do you have any general thoughts on what features to enable (or not enable) in hostapd in order to maximize the chances of successful authentication with clients? If not, any thoughts on why my current workhorse (ALFA AWUS036ACM) authenticates reliably will all devices while my short trial with the AWUS036ACH ended in tears? I'm just wondering if there are general principles related to hostapd and authentication that I am not aware of.

morrownr, 2025-08-16:
G'day,

General Principle no. 1 

Getting the hostapd.conf right depends on the driver in use.

The point being that what works for an out-of-kernel driver may not work for the in-kernel driver for the same chip.

The changes have been coming so fast that some of my documentation needs to be updated. The old drivers for Realtek WiFi 5 need different settings than the new standards compliant in-kernel drivers... at least some settings are different.

The best way to handle this is for me to grab a new sd card for my RasPi4B and build up a new setup for my Alfa ACH. We can do it together and you can see how I determine the correct settings. Would that be okay? If so, it may be a few days before I can start.

My experience is that the Alfa ACM and ACH are two of the best USB adapters ever. They are solid.

morrownr, 2025-08-17:
> you aren't kidding when you put the ACHM in its own separate class ("very long") when it comes to range!

The ACHM is its own class. If a person needs range but throughput is not that important, the ACHM is a very good adapter.

A project that I am working on right now:

I recently bought a new Netgear A9000 USB WiFi adapter. It just came out about a month ago. It uses the mt7925 WiFi 7 chip. The driver has been in the kernel since kernel 6.7 (about a year and half ago). I was testing a patch from a Mediatek dev earlier today... I had reported a throughput problem. The patch is good. I'm getting about 1.2 Gbps throughput with my WiFi 6 router using 5 GHz and 160 Mhz channel width. Fun times.