morrownr/USB-WiFi #645: EDUP WiFi 6e AX3000 wireless USB adapter

owner2gh, 2025-08-20:
Greetings, first I want to thank morrownr for creating such a comprehensive compilation. Now I recently purchase the EDUP WiFi 6e AX3000 wireless USB adapter and installed the suggested firmware. My system is debian 13 trixie, x86_64, 3 USB ports, and one RJ-45.
When I plugged the wifi usb adapter the Lenovo x260 would not recognize the device. Inserted device to different ports, inserted device on 3 other computer usb ports. I finally concluded that perhaps device was defective? (I am not sure).
I would like to know what is the experience of other users of AX3000, since it uses the Mediatek chip set and is mentioned as a recommended wifi adapters.

Thank you again. And thank you for your time.
-Seyed

bjlockie, 2025-08-21:
What model is it?
One is Mediatek, one is Realtek.

bjlockie, 2025-08-21:
I just looked and the specs say both are Mediatek now.
I thought one was Realtek but lsusb will help.
https://www.szedup.com/usb-adaptersBr3/EP1672.html

morrownr, 2025-08-21:
@bjlockie 

> WiFi 6e AX3000

If that is correct, it can only be an adapter with the mt7921au chip as Realtek does not make a WiFi 6e AX3000 chip. They have WiFi 6 AX1800 and WiFi 6e AX5400. However, I would like to see what lsusb has to say.

@owner2gh 

The adapter should not be multi-state which leaves me thinking USB setup should be checked:

Go to the site Main Menu:

https://github.com/morrownr/USB-WiFi

Look under menu item 12 for the following:

USB Ports can be a source of problems with USB WiFi adapters

owner2gh, 2025-08-21:
After receiving the device, I came across the line "USB Ports can be a
source of problems with USB WiFi adapters"... dah... early today I ordered
the same unit again. Thinking perhaps the first wifi adapter was defective.
We'll see. And I will share my experience with you.
Thanks again for all your work and Time you invested to help the community
of freedom respecting people.
Thank you for your time.
-Sed

morrownr, 2025-08-21:
> The model I received is : EP-AX1672

That is the model of the Edup adapter that is in The Plug and Play List. I have one. Very good adapter.

> registered new interface driver mt7921u

That is the correct driver. This means the adapter was detected and the correct driver loaded.

> mt7921u 2-1:1.0: WM Firmware Version: ____010000, Build Time: 20250625153703

The firmware is updated.

> wlxe84e06b0a505: renamed from wlan0

You have a wireless interface.

> $ lsusb

I did not see the adapter with lsusb but it has to be there or we would not see a wireless interface.

> early today I ordered the same unit again. Thinking perhaps the first wifi adapter was defective.

I am confused. You posted enough to convince me that the adapter is working. Did you try to connect to an AP?

> I finally concluded that perhaps device was defective?

Can you elaborate on why you think the adapter is defective? Something must have not worked like you expected... but what? What desktop are you using? KDE, Gnome, other.

bjlockie, 2025-08-22:
Have you tried ALL the USB ports (even the USB2 ones)?

I would boot a different distro from a USB drive.

owner2gh, 2025-08-22:
Yes. I will make a live usb of Ubuntu and see what happens. It is worth a trying.Thanks Sent from my iPad

morrownr, 2025-08-22:
@owner2gh 

> The reason I went astray the vendor readme file directed to download two files
WIFI_MT7961_patch_mcu_1_2_hdr.bin
WIFI_RAM_CODE_MT7961_1.bin
and copy it to /lib/firmware/mediatek folder. From those two files I
associated mediatek firmware as MT7961.

Yes, the chip makers sometimes have families of chips from one core number so you need to be careful when selecting the firmware. Something that can be very handy is the document about firmware that I have on the Main Menu:

https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md

Shallisteria, 2025-08-26:
Maybe try in windows to see if you also have problems there

I also once posted problems with that adapter in the past on both Linux and Windows and returned the unit and then bought a new one (different unit but same model) and actually had no problems with the new ones (same windows and Linux, I even left the Linux untouched) so I'm sure I just got a faulty unit in the past 

I forgot what my problems is but you can just check my post if you're curious enough (currently on phone so it's a hassle to reference you on what I just said)

morrownr, 2025-08-26:
@seth-space 

> WM Firmware Version: ____010000, Build Time: 20241106151045

Your firmware is 2-3 releases old. See No 7 Firmware... at:

https://github.com/morrownr/USB-WiFi

> (about the lar detection by Intel chips)

LAR is an Intel only thing.

Let get this straight, you are using two adapters with mt7921 chips with OpenWRT for the purpose of AP mode with 6 GHz?

If so, tell me your country.

morrownr, 2025-08-26:
@seth-space 

> I am using one EDUP EP-AX1672 Chipset MTK7921AU on a Asus laptop with MX linux. The adapter is recognized by MX linux without any problem but refuses to see the 6GHz SSID's.

Now things are becoming clear.

See option 12 at the Main Menu:

https://github.com/morrownr/USB-WiFi

That is a FAQ.

morrownr, 2025-08-28:
@seth-space 

> So two codes active?

No.

"phy#0 (self-managed)" indicates that your wifi Wi-Fi adapter's regulatory domain (country code) is being managed by the device itself, rather than by your operating system.

> country 00: DFS-UNSET

That means no country code is set in your operating system.

Do you have an Intel wifi card in your system?

You have been trying some things to fix the issue, have you been cleanly removing the fixes if they do not fix the problem?

I do not know what to do next because I do not know the current state of your system.

seth-space, 2025-08-28:
Yes two cards

intel 7265 internal for my Zenbook UX305
and the EDUP usb adapter.

Will look into the changes i made to "fix" things.
Reverting them could be smart indeed.

morrownr, 2025-08-28:
> intel 7265 internal

Sigh. The only time I have seen your results of "iw reg get" are when an Intel card is involved or when I change some code in the Realtek out of kernel drivers (that we are doing our best to retire). 

I think we need to research the effects of LAR to see if having an Intel card can interfere with other cards and adapters. Intel is the only wireless provider that uses LAR and it makes doing AP mode in anything other than 2.4 band basically impossible at least for what most Linux users want to do in AP mode. We might as well act annoyed for 5 minutes, then get over it and figure out what to do.

bjlockie, 2025-08-29:
> LAR is

An Abomination.

morrownr, 2025-09-13:
@seth-space 

I have not run across any good workarounds. There probably is a workaround but I am not running across it.

The last laptop that I bought about 2 years ago came with an Intel wifi card. I do things like AP mode and monitor mode that Intel does not do well, so I bought a little M.2 mt7925 card. It was cheap, does WiFi 7 and the in-kernel driver has been in-place since kernel 6.7. I'm not suggesting that for you as I do not know what kind of slot your system has. You could just remove that Intel card and leave the space open.

seth-space, 2025-09-15:
Replacing the intel is not really an option. Is combined with several ports (hdmi,usb,aux)  on a module. Its not even possible to disable in the bios. So i am stuck with that intel i am afraid.