morrownr/USB-WiFi #649: EDUP EP-AX1672 blocking arp requests?

bdantas, 2025-08-26:
Last night I changed the usb adapter that I use to create the 2.4 GHz AP in my custom dual-band router. I previously used the ALFA AWUS036ACHM. Now I'm using the EDUP EP-AX1672.

Since the switch, I've noticed that wifi clients within the 2.4 GHz subnet cannot ping each other. Problem seems to be that arp requests from clients never get an answer.

Here is the old situation with ALFA AWUS036ACHM creating the AP:
```
$ sudo arping -I wlan0 192.168.20.7
ARPING 192.168.20.7 from 192.168.20.161 wlan0
Unicast reply from 192.168.20.7 [bc:f4:d4:b4:a6:08] 50.663ms
Unicast reply from 192.168.20.7 [bc:f4:d4:b4:a6:08] 5.928ms
Unicast reply from 192.168.20.7 [bc:f4:d4:b4:a6:08] 95.125ms
Unicast reply from 192.168.20.7 [bc:f4:d4:b4:a6:08] 119.184ms
^CSent 4 probe(s) (1 broadcast(s))
Received 4 response(s) (0 request(s), 0 broadcast(s))
```

Here is the new situation with EDUP EP-AX1672 creating the AP:
```
$ sudo arping -I wlan0 192.168.20.7
ARPING 192.168.20.7 from 192.168.20.161 wlan0
^CSent 17 probe(s) (17 broadcast(s))
Received 0 response(s) (0 request(s), 0 broadcast(s))
```
I added `ap_isolate=0` to hostapd.conf but that didn't help (I didn't expect it to help because `ap_isolate=0` is hostapd's default).

Is this a known issue, where the mt7921u's driver prevents ARP requests from traveling between wlan nodes (i.e., stuck in "ap_isolate=1" behavior)? Is there a known fix?

P.S. I saw [this](https://unix.stackexchange.com/questions/645513/clients-cannot-connect-to-each-other-on-wlan-no-reply-on-arp-requests) on stackexchange and it sounds like it is the issue that I'm experiencing.

morrownr, 2025-08-28:
Give me some time to study this and collect information. This problem needs to go to a Mediatek dev but it needs to be a clean report with specific details.

bdantas, 2025-08-29:
Sure thing. To clarify, this is my simple wlan:

(A) ---- (wireless router) ---- (B)

If A and B are wlan nodes on same subnet and A tries to ping B, tcpdump running in router shows me that an arp request arrives at router from A. tcpdump running on B shows that B never receives an arp request, so of course A never receives an arp reply from B. End result is that A's ping request fails with "host not reachable".

If A and B are on different subnets, A can ping B no problem--presumably because clients on different subnets use routing table (rather than arp) to find each other.

So it seems that something in this adapter's latest firmware is preventing a wlan node's arp request from propagating to other wlan nodes.

If you need any additional information from me, just let me know. I'll be happy to help.