morrownr/USB-WiFi #656: [News] Report on the State of USB WiFi on Linux (Sept 2025)

morrownr, 2025-09-15:
Hello Everyone,

Great progress continues regarding USB WiFi on Linux.

Mediatek chips and drivers:

I had two patches accepted for the 6.18 kernel. The patches should go in as work starts in early October. Here is a posting of the USB WiFi relevant patches:

```
mt76 patches for 6.18

Ming Yen Hsieh (3):
       wifi: mt76: mt7925: add MBSSID support
       wifi: mt76: mt7921: add MBSSID support
       wifi: mt76: mt7925: refine the txpower initialization flow

Nick Morrow (2):
       wifi: mt76: mt7925u: Add VID/PID for Netgear A9000
       wifi: mt76: mt7921u: Add VID/PID for Netgear A7500

Quan Zhou (1):
       wifi: mt76: mt7921: Add 160MHz beamformee capability for mt7922 device
```

The Netgear A7500 was reported as not working last week. With some investigating by some knowledgeable users here, it was determined the chip was the mt7921au. A test with a udev rule confirmed this and a new VID/PID was sent in and accepted to go into kernel 6.18. This is interesting as the Netgear A7500 has been available for some time but had not been identified as using a Mediatek chip. It is a lower cost adapter when compared to the A8000 and A9000 (WiFi 7) so I am going to work with the reporter to see if the A7500 should be included in The Plug and Play List. If you have an A7500, let me know.

The patch for the VID/PID for the Netgear A9000 is also going into kernel 6.18. I have one of these adapters and have already added it to The Plug and Play List as a tentative entry until testing is complete. So far, I am impressed with the range and build quality. I did assist a Mediatek dev with a patch that went into 6.17-rc5 that greatly improved speeds. That patch has been backported to 6.16 and 6.12.

Firmware: I noticed new firmware available for the mt7925 as of about two weeks ago.

Realtek WiFi 6/7 chips and drivers:

Driver support  for the rtl8851/33bu and rtl8852/32bu chips went into kernel 6.17 as expected. This support and the drivers, of course, are Linux Standards Compliant (mac80211). Continued work will add USB3 and LED support as well as drivers for the following chips: rtl8852/32cu, rtl8852/32au and rtl8922/12au. If you want to help test or watch the work, see:

https://github.com/morrownr/rtw89

Even though a lot of work is happening to improve the driver experience with Realtek WiFi 6 and 7 devices, I am not removing the warnings against buying Realtek WiFi 6/7 based USB WiFi adapters for now. The reason is that almost all of the Realtek WiFi 6/7 adapters that are on the market right now are multi-state (windows driver onboard). This can and often does cause problems on Linux systems. I'm going to start collecting information about single-state Realtek WiFi 6/7 adapters so if you have single-state (non windows driver onboard) Realtek WiFi 6/7 adapters, please let me know.

Until next time,

@morrownr

morrownr, 2025-09-15:
I forgot to mention:

Now that we have mac80211, in-kernel drivers for the rtl8814au, rtl8812au and rtl8821/11au chips, I am making new sections and am adding known good adapters to The Plug and Play List.

a-gavin, 2025-11-18:
Good to know, thanks for the info! I know this is a USB NIC-focused project, but I also saw issues with my laptop's built-in mt7921 chipset radio, given that this is common code for all the mt792x drivers (afaik). Glad things are resolving okay

morrownr, 2025-11-18:
@a-gavin 

Good information to know.

Your adapter is listed in The Plug and Play List but I do not have one. It is listed based on information from others. Nobody has done a review. If you do a review, I'll be glad to post it in The Plug and Play List with the adapter.

a-gavin, 2025-11-18:
Hopefully not off topic for this issue, but what would a review entail @morrownr?

I've used the NIC in STA, AP, and monitor mode. Save this regression, it's been pretty good. I work with someone who uses several of them running off a USB hub without issues as well, primarily in STA mode.