morrownr/USB-WiFi #657: [Help]: mt7921u can only connect to other wifi 6e devices

chcoomilk, 2025-09-16:
### Checklist

- [x] I acknowledge that support is provided on a best-effort basis.
- [x] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

Linux chocomini 6.14.0-29-generic #29-Ubuntu SMP PREEMPT_DYNAMIC Thu Aug  7 18:32:38 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux

### lsusb

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 001 Device 002: ID 0bda:b85b Realtek Semiconductor Corp. Bluetooth Radio Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 002 Device 003: ID 7825:a2a4 Other World Computing External SATA Hard Drive Adapter cable PA023U3 Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 003 Device 002: ID 320f:5088 Telink Wireless Gaming Keyboard Bus 003 Device 003: ID 0573:1573 Zoran Co. Personal Media Division (Nogatech) USB Audio and HID Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 004 Device 003: ID 0e8d:7961 MediaTek Inc. Wireless_Device

### rfkill

0: hci0: Bluetooth         Soft blocked: no         Hard blocked: no 1: phy0: Wireless LAN         Soft blocked: no         Hard blocked: no 3: phy2: Wireless LAN         Soft blocked: no         Hard blocked: no

### dkms

```text
none (I didn't have dkms before this, do I need to reboot after installing?)
```

### iw

```text
phy#2
        Unnamed/non-netdev interface
                wdev 0x200000002
                addr 92:de:80:83:9d:e0
                type P2P-device
                txpower 3.00 dBm
        Interface wlx90de80839de0
                ifindex 15
                wdev 0x200000001
                addr 90:de:80:83:9d:e0
                type managed
                txpower 3.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       0       0       0       0       0       0               0
phy#0
        Interface wlp3s0
                ifindex 3
                wdev 0x1
                addr c0:e3:50:cf:14:44
                type managed
                txpower 0.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       0       0       0       0       0       0               0
```

### What happened?

I ordered this generic wifi 6 usb, came with a small cd which I'm assuming is for windows driver install. The wifi dongle works fine out of the box without having to compile anything.

I have a wifi 5 router at home, it's working fine with other devices and it's picked up by the new wifi dongle but unable to connect to it. If I try connecting with nmcli `sudo nmcli device wifi connect HomeRouter --ask`, I'll only get:

```
Push of the WPS button on the router or a password is required to access the wireless network 'HomeRouter'.
Password (802-11-wireless-security.psk): **********
Error: Connection activation failed: The Wi-Fi network could not be found.
```

I tried connecting with 3 other different type of wifi hotspot
wifi 6: fail
wifi 6e: success
wifi 7: fail

here's the iw dev output if connected to my 6e device hotspot (it's a google pixel 6 pro)

```
phy#2
        Unnamed/non-netdev interface
                wdev 0x200000002
                addr 92:de:80:83:9d:e0
                type P2P-device
                txpower 3.00 dBm
        Interface wlx90de80839de0
                ifindex 15
                wdev 0x200000001
                addr 90:de:80:83:9d:e0
                ssid three5495
                type managed
                channel 161 (5805 MHz), width: 80 MHz, center1: 5775 MHz
                txpower 3.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       0       0       0       0       0       0               0
phy#0
        Interface wlp3s0
                ifindex 3
                wdev 0x1
                addr c0:e3:50:cf:14:44
                type managed
                txpower 0.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       0       0       0       0       0       0               0
```

chcoomilk, 2025-09-18:
it appears that this is most likely a hardware problem. I've tried replacing the driver with the latest, different usb ports, and even tried installing it on windows, the problem stays the same. It picked up nearby wifi perfectly fine, but just won't connect to it. 

The seller is giving me a hard time, hopefully I can get my money back.

chcoomilk, 2025-09-18:
hey, thank you for reaching out

> I would be interested in a link to the product if you care to share.

here's the exact [product page](https://www.tokopedia.com/buanaonlinestore/rkusb-wifi-dual-band-wireless-2-4ghz-5ghz-wifi-6-internet-network-adapter-1730765965302990527) where I bought it. It might be a little bit hard to navigate around the page because it's not in english, you'll need a bit of google translate, but more or less here is the description bit of the product of interest in english

```
USB Wifi Dual Band AX1800 Wifi 6 Description
* 100% Brand New Product
* MediaTek Chipset
* Dual Band 2,4Ghz and 5Ghz Wifi
* Super Speed USB 3.0
* Support Wifi 6
* Speed up to 1800Mbps
* Support MU-MIMO dan OFDM
* Support Windows 10 / 11
```

the packaging and the product are very similar if not the same to what this person has https://github.com/morrownr/USB-WiFi/discussions/480#discussioncomment-12671093

> I am not familiar with this Linux distro

it's good ol' Ubuntu 25.04 and on wayland if that's of any relevance

> iw reg get

```
country ID: DFS-JP
        (2400 - 2483 @ 40), (N/A, 26), (N/A), NO-OUTDOOR
        (5150 - 5350 @ 80), (N/A, 23), (N/A), NO-OUTDOOR
        (5725 - 5825 @ 80), (N/A, 23), (N/A), NO-OUTDOOR
```

>The adapter seem to be connected. I see that you have an internal card. Is that an Intel based card?

no, it's realtek

```
lspci

...
03:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8852BE PCIe 802.11ax Wireless Network Controller
...
```

chcoomilk, 2025-09-18:
> I avoid listing generic (whitebox) adapters in The Plug and Play List

I've seen the list, and I've also tried finding any wifi 5 to 7 products sold in my country from that list. Unfortunately, they're either not sold here or cost 5-10x more, so I've basically resorted to playing gachapon usb wifi dongle

bjlockie, 2025-09-19:
Please tell me "passive scan" is not that Intel abomination.

dubhater, 2025-09-19:
> the packaging and the product are very similar if not the same to what this person has [#480 (comment)](https://github.com/morrownr/USB-WiFi/discussions/480#discussioncomment-12671093)

I have one that looks just like that, but inside is the RTL8832BU chip.

chcoomilk, 2025-09-20:
> What are the details of the "WiFi 5" router? 

[image]

https://openwrt.org/inbox/toh/xiaomi/xiaomi_mi_router_4a_gigabit_edition#info (link is for my router's hardware info, I'm not using openwrt.. yet)

I'll try setting the country to US then, see if it changes anything

chcoomilk, 2025-09-20:
> Please tell me "passive scan" is not that Intel abomination.

hey, as long as it works, unlike mine :/

chcoomilk, 2025-09-20:
hi @morrownr 

I've tried everything I could, I changed my Wifi 5 router settings a bunch of times mixing all sorts of configurations and no luck. Currently, it is set to channel 36, 20 mhz bandwidth, WPA/WPA2. I'll set it back to just WPA2 sometime later.

But changing country code to the US (`sudo iw reg set US`) did yield a good result. Now I can connect to one of my other phones, which is wifi 6, not the 6e one, that I'll be using as a hotspot for now.

> To help decide whch channel may be the best channel to use, you can install an app called WiFi Analyzer

I'm already using WiFiman from Ubiquiti prior for scanning to setup my wifi router. Every channel is green because there's little interference and no one is using 5ghz in the area.

> While I am not fully versed in wifi usage in Japan, I think some things that I know could help. In fact, I just asked AI and here is what it said:

I myself have the slightest to no idea why it's printing out `DFS-JP`. I do not live in Japan, but I'm guessing we're using the same regulation I suppose.

Thank you for all your time and effort @morrownr, but I'd feel bad to continue diagnosing this issue while having another strange hardware problem until I figure out what's going on with the only usable 2 usb ports of the computer. Basically, in some rare occasions, lsusb will show the wifi adapter is connected, but it wouldn't appear in iw dev. I need to slightly push the body of the usb downward so I can get the wifi adapter to appear in iw dev. But judging by the number (phy#56) printed by running `iw dev`, this means the usb has been reconnected multiple times while already inserted. This could explain why the previous dongle died. Oddly, this only happens if it's a wifi adapter. I can connect usb storage or my phone just fine.

closed 2025-09-20 by chcoomilk

morrownr, 2025-09-20:
@chcoomilk 

So, you are not in Japan. That means your system is using an incorrect country code. The adapter may be providing  internal information that your distro is using. The adapter may have been designed to be sold in Japan. I have a FAQ on the site Main Menu:

https://github.com/morrownr/USB-WiFi

Go down to menu item 12. The first item in the FAQ could help you fix this. Let me know if this helps.

chcoomilk, 2025-09-20:
I tried adding `options cfg80211 ieee80211_regdom=ID` to cfg80211.conf and rebooted, still nothing. And 1 thing to note is my country has always been set correctly

when you asked me to run iw reg get
https://github.com/morrownr/USB-WiFi/issues/657#issuecomment-3308099880

> country ID: DFS-JP

my country is ID as in Indonesia, but the DFS-JP is something a user cannot manually change I don't think

chcoomilk, 2025-09-20:
oh and I expected the wifi adapter to fail connecting again because I changed my `iw set` back to ID from US, but I'm able to connect, so something else must have fixed the issue, not setting the country code, sorry

morrownr, 2025-09-20:
> my country is ID as in Indonesia, but the DFS-JP is something a user cannot manually change I don't think

I read that wrong so my bad. 

> so something else must have fixed the issue

It could be the channel change if you did indeed change the router channel. On the other hand, WiFi is complicated.